Serasidis/STM32_HID_Bootloader #1: upload does not work at all on STM32F407ZET6

uzi18, 2018-09-03:
I'm trying to use this bootloader but always stops on sending first 1024bytes:
```

> Searching for 1209:BEBA HID device...
#
> 1209:BEBA device is found !
> Sending <reset pages> command...
> Flashing firmware...
. 1024 Bytes
^C

```

any idea what is wrong?

uzi18, 2018-09-06:
it is nothing you know -  PLD Linux, but I dont think it is OS problem. 
need to debug if any data are realy  received by hid bootloader

also found bug in cli program but this is for another issue

uzi18, 2018-09-12:
did some checks, program is waiting for acknowledgement from bootloade, about 1kB already flashed, however hid report is not sent or not received.

page reset raport is sent by hid-flash and received by bootloader - this works

next step: hid-flash sends 1024B block but not receive info to send next block

@serasidis any idea?

Serasidis, 2018-09-15:
@uzi18 I suppose:
1. you use the [Arduino_STM32 ](https://github.com/rogerclarkmelbourne/Arduino_STM32) repository (_...arduino_folder/hardware/Arduino_STM32_)
2. You have applied the [patch](https://github.com/Serasidis/STM32_HID_Bootloader/blob/master/Arduino_STM32_patch_180809_02.zip)
3. You have the BOOT-1 and BOOT-2 pins connected to the GND

I have successfully tested the **HID-BL 2.0** to 2 [Diymore STM32F4 Discovery board](https://www.amazon.co.uk/Diymore-Discovery-STM32F407VGT6-Microcontroller-Development/dp/B0788MVHVT) (407VGT6) and [ArchMAX 1.1](http://wiki.seeedstudio.com/Arch_Max_v1.1) (407VET6)

Which STM32F4 board do you use ?

uzi18, 2018-09-15:
this board: https://www.solidrop.net/product/stm32f407zet6-development-board-m4-stm32f4-core-board-arm-development-board-cortex-m4.html

have bypassed any check to go into bootloader flash mode directly, but it does not work as I expected. 
sending reports from cpu to os hid-flash works but some data packets from hid-flash are lost.
need to check on another module or/and if usb/cpu is correctly soldered.

uzi18, 2018-09-15:
I think it is needed to add some sync every 64B received, if timeout reached resend it one more time.

uzi18, 2018-09-15:
@Serasidis changed board and have some progress here, will test it one more time without my modifications. Also need to find what is wrong with other board.

Serasidis, 2018-09-24:
@uzi18 is there any progress ?

BennehBoy, 2019-01-20:
@Serasidis shout if this needs a separate issue, but on Black407VET6 (and presumably f4xx) first upload works fine, but upon system reset the board is in perpetual bootloader, am I doing something wrong?

BennehBoy, 2019-01-21:
Oops - had BOOT0 set high, removed jumper and now working.

Have you tested against a blackZET6 board?

uzi18, 2019-01-21:
Will try later and give you feedback

BennehBoy, 2019-01-21:
Actually I think BOOT0 may be a red herring, it only occasionally goes directly to sketch execution - perhaps due to it floating?

BennehBoy, 2019-01-21:
OK, so got to the root of the issue, the code, although it has a comment in saying 407VET6, has the incorrect GPIO's set for BOOT1 & the LED.

After setting this up correctly for Black407VET6 & ZET6, with BOOT1 high & BOOT0 low, both cards boot straight into the sketch.  Sorry for all the noise in here, but got there in the end.

Serasidis, 2019-06-11:
The **STM32 HID Bootloader** is available on the Official [Arduino_Core_STM32](https://github.com/stm32duino/Arduino_Core_STM32) Core

closed 2019-06-11 by Serasidis