Serasidis/STM32_HID_Bootloader #12: Error running on Linux (32 and 64 bits)

rtek1000, 2019-04-21:
Hi,

The previous version was compiling but was not uploading.

The current version is no longer compiling.
[image]

But SAM is ok.

Linux: Manjaro (XFCE) 32 bits and 64 bits

(ST-link V2 works, does not seem to be a problem with the system, both are newly installed)

Note: In the previous version was showing 'HID bootloader V3.0', and in the current is 'V2.0'.

Thank you.

Screenshots:
[32 bits.zip](https://github.com/Serasidis/STM32_HID_Bootloader/files/3100923/32.bits.zip)
[64 bits.zip](https://github.com/Serasidis/STM32_HID_Bootloader/files/3100924/64.bits.zip)

rtek1000, 2019-04-21:
Hi,

Note: To write the bootloader it is necessary to perform the complete erase of the chip.

(The st-link GUI that Manjaro Linux installed did not have the erase option, so I had to do it via the command line)

On the console, the lsusb command showed the old VID: PID (1eaf: 0004), after burning the new bootloader (1209: beba) without full erase.

With the erase (st-flash erase), the lsusb command showed '1209: beba' correctly, after burning the new bootloader.

Thank you.

uzi18, 2019-04-21:
@rtek1000 I think you just need to force to not leave hid bootloader with jumper on dedicated pin

rtek1000, 2019-04-21:
Hi,

Thanks, but I do not think it's a jumper problem,

The current problem is to compile, if I change some files, such as boards.txt / platform.txt then the IDE can compile.

If I download the STM32 files and overlap the files with the patch files (from this repository), then it can not compile. The files have been updated recently, and the ones before being updated that I downloaded were compiling.

(This source: https://github.com/rogerclarkmelbourne/Arduino_STM32)

I tried with the source Arduino_Core_STM32, but the files are saved otherwise by the Boards Manager, and cannot works.

rtek1000, 2019-04-21:
Hi,

I was able to do compile, it seems to be some optimization configuration that does not let compile.

By returning these lines, it seems to be able to compile without errors, but the upload part has not yet worked.

[image]

rtek1000, 2019-04-21:
One thing I noticed is that these patch files are older than the current ones from the other repository (https://github.com/rogerclarkmelbourne/Arduino_STM32)

Thank you.

Serasidis, 2019-06-11:
The **STM32 HID Bootloader** is available on the Official [Arduino_Core_STM32](https://github.com/stm32duino/Arduino_Core_STM32) Core

closed 2019-06-11 by Serasidis