Serasidis/STM32_HID_Bootloader #18: STM Official Core Support

wanders78746, 2019-07-23:
I apologize in advance, as I am a n00b just getting started in STM32duino development, but I've decided to use the STM Official Core. Will there be support for it soon? THANKS for the impressive work so far!

fpistm, 2019-07-23:
This is already supported. 😉

wanders78746, 2019-07-23:
Sorry, it wasn't clear from the descriptioni. Thanks! Nothing like a 2K bootloader that doesn't need a driver in Windows.

fpistm, 2019-07-23:
No worry ;)
Here a note by @Serasidis in the Wiki:
https://github.com/stm32duino/wiki/wiki/Upload-methods#hid-bootloader-22-hid-bl

Andy2No, 2019-08-25:
The Readme page still says 
> 8. STM Official Core will be supported soon.

Is that just an out of date comment?  See:

https://github.com/Serasidis/STM32_HID_Bootloader/blob/master/README.md

Serasidis, 2019-08-26:
Out of date information has been removed from README.md file

closed 2019-08-26 by Serasidis