Serasidis/STM32_HID_Bootloader #19: USB device not regonised, no PORT available, link to patch is broken

joej970, 2019-08-21:
After flashing the bootloader I cannot execute step where the patch has to be applied because the link is broken. 
Furthermore when I connect blue pill directly (not via FTDI dongle), computer cannot recognise the device and there are no available ports in Arduino IDE under tools > port.
Arduino IDE return error: Error - [1209:BEBA] device is not found
I am using [Arduino_Core_STM32](https://github.com/stm32duino/Arduino_Core_STM32)

khyarul, 2019-10-24:
it also happens to me, after first upload, my computer can't recognise the HID device, I'm using maple mini

joej970, 2019-11-01:
Try setting USB support to _CDC (generic serial supersede USART)_

toogooda, 2019-12-01:
> it also happens to me, after first upload, my computer can't recognise the HID device until I flash again the bootloader and so on, I'm using maple mini

Same issue here, requires reflash of bootloader after every use

joej970, 2019-12-01:
It happened to me when I was using MPU 9250 DMP library that crashed  STM32F1 because code started to execute too fast. My fix was to add delay(5000) before calling init() function of my library. The cause was probably that external sensor (MPU 9250) was not ready quick enough and this hanged STM32F1.

Now I use HID bootloader 2.2 (and delay) and it works fine for most of the time (I still have to unplug and plug stm32f1 some times, but I do not have to reflash it) . I have USB support set to CDC.

fpistm, 2019-12-02:
Hi,
Just a quick note, to ensure Serial is well initialized before use them add:
`while(!Serial); `
after `Serial.begin();`.
See:
https://www.arduino.cc/reference/en/language/functions/communication/serial/ifserial/