Serasidis/STM32_HID_Bootloader #20: Link to patch file for Roger core no longer works

raghur, 2019-08-26:
> Apply the Arduino_STM32_patch from here in case of using Roger's Core (select the zip file according to your Operating System).

Link to patch is broken... I got it from github history - but is there a reason to not provide it/support it?

Serasidis, 2019-08-28:
Version 2.1.0 includes the patch files for Win, Linux and MacOS

https://github.com/Serasidis/STM32_HID_Bootloader/releases/tag/2.1.0

raghur, 2019-09-02:
I picked them up from git history... but I don't understand why they were deleted in the first place?

Serasidis, 2019-09-03:
It is better to separate the source code from the binary files. Those who just want to program the Bootloader do not need to download the source code.

raghur, 2019-09-06:
I understand... perhaps you can host it on Gdrive or something and link to it from the README if you don't expect source changes to the IDE patch. 

Feel free to close the issue.

closed 2019-09-10 by Serasidis