Serasidis/STM32_HID_Bootloader #23: Nothing on USB after Update Hid Bootloader Stm32F407CC Using Stm32Cube Thru StlinkV2

ajaybnl, 2019-11-02:
First, My New BlackPill Stm32F407CC Talks via USB Serial Com Port (USB WORKS).

I Downloaded compiled version HID Bootloader 2.2 (from this site) for F4xx 

I tried HID Bootloader Flashed to  Default 0x00 (it shows 0x08 in menu)  Using Stm32Cube Thru StlinkV2 (Updated firmware first)

Success, 

**But no USB port shown in device manager when i connect BlackPill F407CC From USB.**

Also , Please suggest **What is  [1209:BEBA] Device.**

**BTW:** I successfully uploaded the arduino code from stlinkv2 (updated firmware)  (LED on D17) 
But i want to upload with only USB.

I have Blue pill f103 , Daplink (l11u35), stlinkv2 devices.

roboticboyer, 2020-07-18:
I suggest to use the [WeAct Bootloader](https://github.com/WeActTC/MiniF4-STM32F4x1/tree/master/Soft/WeAct_HID_FW_Bootloader)

There is also a fork the [HID Bootloader](https://github.com/WeActTC/STM32_HID_Bootloader/releases) from Serasidis' one

samveen, 2020-08-06:
@roboticboyer The WeAct Bootloader doesn't seem to integrate with Arduino IDE like the HID Bootloader does. Not only that, their fork of the HID Bootloader modifies the flag check, and thus also doesn't work with the HID Bootloader upload method in the Arduino IDE. I managed to get a [working version](https://github.com/samveen/STM32_HID_Bootloader/tree/merged), but I am unsure if it qualifies to be be merged into here.

roboticboyer, 2020-08-06:
I wrote some [notes](https://hardwareliberopinerolo.github.io/site/blackpill/) how to use the  WeAct Bootloader
@samveen where is available your bin file of the modified bootloader?

samveen, 2020-08-07:
@roboticboyer I didn't put it up anywhere, expecting that people might be more comfortable building it themselves. However you can find it via [this google drive link](https://drive.google.com/file/d/1-CE08Ry0gNKHBgiqrhTOt3GIq34PNQR_/view?usp=sharing)

samveen, 2020-08-28:
I used `dfu-util` to write it to the board. One thing I'll note is that I'm using a `411`, instead of a `401`. May I suggest that you look at the code and see if you can build it locally instead of using the one I built, given that I am a beginner myself, so my changes might need reviewing before you use them.

roboticboyer, 2020-08-29:
Thank you
I'll check and recompile the firmware for the 401 
Please only confirm that the parameters of dfu-util are correct:
`./dfu-util -a 0 -s 0x08000000 -D hid_bootloader.bin`

samveen, 2020-08-29:
I used `dfu-util -a 0 -d 0483:df11 -s 0x8000000:mass-erase:force -D hid_bootloader.bin -t 1024`

zoobab, 2021-02-15:
I have a WeAct BlackPill:

http://zoobab.wikidot.com/stm32f401-devboard

Those guys at WeAct are shooting themselves in the foot, as what people are interested in is Arduino IDE support, which cannot happen if the flashing tools are forked from BSD-licensed software and made proprietary.

dreamcat4, 2021-04-12:
hello, sorry am a bit new to this device, and just wanted to know:

* why is hid bootloader better than dfu mode?
* how can i upload sketches without having to press a physical button on the device, etc?

rgoulter, 2021-04-12:
> how can i upload sketches without having to press a physical button on the device, etc?

A compilation output of an Arduino sketch is a firmware for the device. A bootloader is firmware which helps you update the 'application' firmware (e.g. the Arduino sketch).

The easiest / most reliable way I've found is to use an STLink.

Whereas, I've found DFU to be ... finicky; sometimes only working if I hold down both the BOOT and NRST buttons for a long time (30s+), or not work at all. (Different levels of success depending on the OS / which particular dev board is used).

This HID bootloader in this repo (IIUC) would allow updating the firmware when used with the hid flashing program (also in this repo). This particular fork of the HID bootloader is geared towards Arduino. -- The benefit being, you could update the firmware without needing to push any buttons on the dev board. - If you flash the bootloader with the correct configuration, this should let you upload from Arduino IDE.

dreamcat4, 2021-04-12:
well i have the weact f411 here on platformio. and have been flashing it in dfu mode. pressing the button

could plugin an stlink i suppose. its just more wires / cables

but if you think i can have hid flashing without needing the buttons for this device? because that might seem worth it

can then add a custom uploader cmdline in the platformio.ini file to call it

however what confuses me, and makes me ask about butonless HID mode is because the README seems to say i still need to press some button... key instead of boot0, same difference really.

in which case it does not achieve the set goal, and is no better than dfu mode for me. sorry for the confusion, but i am really confused by this. wherever i look online about it for this particular board :)

hehe

dreamcat4, 2021-04-12:
in the WeAct one here

https://github.com/WeActTC/MiniSTM32F4x1/tree/master/Soft/WeAct_HID_FW_Bootloader

rgoulter, 2021-04-12:
Ah, that's right. Sorry, I haven't yet tried this bootloader. I recall seeing the code has a comment in the entrypoint:

```
In case of incoming magic number or <BOOT_1_PIN> is LOW, jump to HID bootloader
```
https://github.com/Serasidis/STM32_HID_Bootloader/blob/master/bootloader/F4/Src/main.c#L138
(Per WeAct's changes, the WeAct MiniF4's `<key>` is the `boot_1` being referred to).

so presumably, the bootloader can be entered by setting the value & triggering a reset from the firmware, and could be entered without pushing a button.

dreamcat4, 2021-04-12:
> and could be entered without pushing a button.

how?

you lost me again. sorry i could not follow that explanation

rgoulter, 2021-04-12:
This is what makes sense to me; I haven't tried it. I don't have any significant experience with embedded development; so I'd suggest the search terms "stm32 software reset"; and/or figure out how to set a particular register to a particular value in the stm32duino port.

Again, the bootloader in this repo enters the HID bootloader if the boot_1 is low (i.e. `<key>` is held down), or "incoming magic number" (value of a particular register has a specific value) when the bootloader starts. i.e. after a reset or when the device is plugged in. - As you point out, the "key held down" part isn't more useful than the built-in DFU; but I'd expect it's possible to write to a register and reset the MCU from the firmware.

dreamcat4, 2021-04-12:
so this is... some WIP not finished yet?

https://github.com/stm32duino/Arduino_Core_STM32/issues/1118#issuecomment-657580833

in PR  #710 ?

pathnirvana, 2021-04-24:
> I wrote some [notes](https://hardwareliberopinerolo.github.io/site/blackpill/) how to use the WeAct Bootloader
> @samveen where is available your bin file of the modified bootloader?

I was able to get f401cc working with platformio by following some of the above steps by @roboticboyer 

In wsl or linux
`
objcopy -Iihex -j .sec1 -Obinary WeAct_HID_Bootloader_F4x1.hex WeAct_HID_Bootloader_F4x1.bin
dfu-suffix -a WeAct_HID_Bootloader_F4x1.bin 
`

then in windows
`
.\st-flash.exe write .\WeAct_HID_Bootloader_F4x1.bin 0x8000000
`

finally in platformio.ini file
`
[env:blackpill_f401cc]
platform = ststm32
board = blackpill_f401cc
framework = arduino
upload_protocol = dfu
`

to put in the bootloader mode press nrst button while boot0 is pressed. you should see STM32 BOOTLOADER in windows device manager when it is in the bootloader mode.

hope this helps someone

pathnirvana, 2021-07-26:
probably not much related here. But recently I changed to using tinyuf2 instead of the hid bootloader for both f401 and f411 blackpills. much easier to get into the bootloader mode - just double tap the reset button.
can be flashed easily with 
`.\st-flash.exe write .\tinyuf2-stm32f401_blackpill-0.5.0.bin 0x8000000`

also the releases can be downloaded from https://github.com/adafruit/tinyuf2/releases/

xromansx, 2022-09-14:
So for now only one working option to flash blackpill through USB without touching it is WeAct_HID_Bootloader

xromansx, 2022-09-22:
> I used `dfu-util` to write it to the board. One thing I'll note is that I'm using a `411`, instead of a `401`. May I suggest that you look at the [changed code](https://github.com/samveen/STM32_HID_Bootloader/tree/merged) and see if you can build it locally instead of using the one I built, given that I am a beginner myself, so my changes might need reviewing before you use them.

I recompiled you version but can't get it working.

samveen, 2022-09-30:
> I recompiled you version but can't get it working.

It's been quite some time since I touched it. Things may have changed a lot during that time. Please check the errors and see if you are able to fix them, as I haven't touched this code for more than 2 years.

closed 2022-11-20 by ajaybnl

fufu930, 2024-12-18:
https://github.com/WeActStudio/WeAct_HID_Bootloader_F4x1/issues/2#issuecomment-2550300306