Serasidis/STM32_HID_Bootloader #24: An error occurred while uploading the sketch

mrguen, 2019-11-26:
Hello,

I just flashed my bluepill with hid_generic_pc13.bin. It seemed to flash properly. But I can't load a sketh. Can there be a verbose output?

---------------------------------------------------------------------------------------------
Flashing the bluepill
`
C:\temp\stlink-1.3.0-win64\stlink-1.3.0-win64\bin>st-flash.exe write hid_generic
_pc13.bin 0x8000000
st-flash 1.3.0
2019-11-26T11:27:56 INFO C:\Users\Jerry\Desktop\stlink-master\src\usb.c: -- exit
_dfu_mode
2019-11-26T11:27:56 INFO C:\Users\Jerry\Desktop\stlink-master\src\common.c: Load
ing device parameters....
2019-11-26T11:27:56 INFO C:\Users\Jerry\Desktop\stlink-master\src\common.c: Devi
ce connected is: F1 Medium-density device, id 0x20036410
2019-11-26T11:27:56 INFO C:\Users\Jerry\Desktop\stlink-master\src\common.c: SRAM
 size: 0x5000 bytes (20 KiB), Flash: 0x10000 bytes (64 KiB) in pages of 1024 byt
es
2019-11-26T11:27:56 INFO C:\Users\Jerry\Desktop\stlink-master\src\common.c: Atte
mpting to write 1976 (0x7b8) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08000400 erased
2019-11-26T11:27:56 INFO C:\Users\Jerry\Desktop\stlink-master\src\common.c: Fini
shed erasing 2 pages of 1024 (0x400) bytes
2019-11-26T11:27:56 INFO C:\Users\Jerry\Desktop\stlink-master\src\common.c: Star
ting Flash write for VL/F0/F3 core id
2019-11-26T11:27:56 INFO C:\Users\Jerry\Desktop\stlink-master\src\flash_loader.c
: Successfully loaded flash loader in sram
  1/1 pages written
2019-11-26T11:27:57 INFO C:\Users\Jerry\Desktop\stlink-master\src\common.c: Star
ting verification of write complete
2019-11-26T11:27:57 INFO C:\Users\Jerry\Desktop\stlink-master\src\common.c: Flas
h written and verified! jolly good!

C:\temp\stlink-1.3.0-win64\stlink-1.3.0-win64\bin>`

---------------------------------------------------------------------------------------------
Trying to load blink.ino

+-----------------------------------------------------------------------+
|         HID-Flash v2.2.1 - STM32 HID Bootloader Flash Tool            |
|     (c)      2018 - Bruno Freitas       http://www.brunofreitas.com   |
|     (c) 2018-2019 - Vassilis Serasidis  https://www.serasidis.gr      |
|   Customized for STM32duino ecosystem   https://www.stm32duino.com    |
+-----------------------------------------------------------------------+

Usage: hid-flash <bin_firmware_file> <comport> <delay (optional)>
An error occurred while uploading the sketch

matthijskooijman, 2020-03-25:
You should probably enable verbose uploads (and maybe also compilation) in the Arduino IDE, if you're using that, which should show the exact hid-flash commandline that is used. Likely it is being generated wrongly. What STM32 Arduino core are you using?

MatthewReed303, 2020-06-10:
I'm also having the same error and have verbose enabled and provides no extra details. here is the link to the forum post I created https://www.stm32duino.com/viewtopic.php?f=58&t=489&p=3264#p3264

the-alchemist88, 2020-07-02:
+-----------------------------------------------------------------------+
|         HID-Flash v2.2.1 - STM32 HID Bootloader Flash Tool            |
|     (c)      2018 - Bruno Freitas       http://www.brunofreitas.com   |
|     (c) 2018-2019 - Vassilis Serasidis  https://www.serasidis.gr      |
|   Customized for STM32duino ecosystem   https://www.stm32duino.com    |
+-----------------------------------------------------------------------+

Usage: hid-flash <bin_firmware_file> <comport> <delay (optional)>

Looks smilar. Still no solution? Kind of newbie here :)

jlarnal, 2020-08-07:
Same here. Selected device is _BluePill F130C8_, sketch is the "_Blink_" example, to test it out.
Compiler & Uploader verbose are both enabled.

[image]

flaviohpo, 2020-08-14:
I had the same problem.

jerkey, 2023-07-21:
this is because Arduino is not invoking the hid-flash command correctly, it's mandatory to supply a second parameter after the filename.  If you look at the line just before the red text, hid-flash is invoked with only the filename.

if(argc < 3) { // from hid-flash/main.c
    printf("Usage: hid-flash <bin_firmware_file> <comport> <delay (optional)>\n");
    return 1; }

My device is on USB so there is no "comport" but I have to put some text there to get past this error, but then it finds the USB device.