Serasidis/STM32_HID_Bootloader #25: Serial via USB

toogooda, 2019-11-30:
Got it to upload but then com port disappears  is there supposed to be serial support when using this boot-loader?

ssieb, 2019-11-30:
It uses the HID interface, not serial.

toogooda, 2019-11-30:
> It uses the HID interface, not serial.

Yes thanks, my question is does that then mean I can't get my serial debug output via the usb because it is being used as a HID or is this part of the bug that then stops all other uploads after the first one?

ssieb, 2019-11-30:
If you use this bootloader, then you need to use the HID uploader.  Once your sketch is running, it will enable the serial port.
(This is my recollection, I haven't actually used this device for a while.)

toogooda, 2019-11-30:
> If you use this bootloader, then you need to use the HID uploader. Once your sketch is running, it will enable the serial port.
> (This is my recollection, I haven't actually used this device for a while.)

Great thank you that is what I wanted to know. I am using the ST uploader HID option so there seems to be a bug with it. As it does the upload but then the COM port fails and there is no serial output.

toogooda, 2020-02-10:
Is this still active? great idea but looks like its not actively being worked on.

uzi18, 2020-02-10:
what core did you use?
it work so what you want to add/change?

toogooda, 2020-02-11:
> what core did you use?
> it work so what you want to add/change?
Upload worked but then no serial is possible which is a show stopper.
Tried it in multiple devices.

uzi18, 2020-02-11:
ok but what core you use for it?

toogooda, 2020-02-11:
> ok but what core you use for it?

Offical ST one think out is called STM32duino

uzi18, 2020-02-11:
version?

toogooda, 2020-02-11:
AT work so not sure sorry will update tonight. Its 9am here :)

uzi18, 2020-02-11:
10pm here and afterwork :)