Serasidis/STM32_HID_Bootloader #26: No HID device found

skandragon, 2019-11-30:
I flashed `hid_generic_pc13.bin` to several of my devices, but I am unable to see the HID device on the USB bus.  Arduino claims:

```
-----------------------------------------------------------------------+
|         HID-Flash v2.2.1 - STM32 HID Bootloader Flash Tool            |
|     (c)      2018 - Bruno Freitas       http://www.brunofreitas.com   |
|     (c) 2018-2019 - Vassilis Serasidis  https://www.serasidis.gr      |
|   Customized for STM32duino ecosystem   https://www.stm32duino.com    |
+-----------------------------------------------------------------------+

> Trying to open the [cu.usbserial-A5018LV7]...
> Toggling DTR...
> Searching for [1209:BEBA] device...
##########
An error occurred while uploading the sketch
Error - [1209:BEBA] device is not found :(> Searching for [cu.usbserial-A5018LV7] ...
> [cu.usbserial-A5018LV7] is found !
> Finish
```

I've used many uP before, but this one seems to be giving me problems.  Any idea what I am doing wrong here?

I have a FTDI connected, but am not sure where DTR is supposed to be connected.  This may be part of the issue.  I am powering only over USB, which is also connected of course.

Thanks!

uzi18, 2019-11-30:
what is your board/cpu ?

stevstrong, 2020-04-12:
I have the same problem with the bluepill board.

Flashed with STLink utility the latest generic pc13 bin file.
Then pressed reset, nothing happened.
When unplug ->replug the USB cable, Windows device manager recognizes it as 
`HID compliant vendor-defined device.`
Upload with Arduino a first blinky sketch -> OK.

Next try to upload with Arduino hangs after the first 1024 bytes.
Board reset, nothing happens. USB cable unplug - replug.
Windows device manager reports `Unknown USB device (Device Descriptor Request Failed)`
Tried to set BOOT0 jumper to high, no change.
In this state no more upload with Arduino possible.

Flashing my custom BIN via STLink utlity -> works.
Flashing HID bootloader with STLInk -> works, device manager recognizes it as HID device.
But any further upload fails as the VID:PID cannot be found.

carbonadam, 2020-06-11:
Pretty sure I have a genuine blue pill but after successfully flashing it(fast blinking led) its not showing up in windows.. its a 68 with 128kb which is either the high density or low density right?  Not sure whats up

Riffer, 2020-08-08:
@stevstrong I beleave that 2.2.2 bootloader and core STM32 1.9 are not compatible. After downgrading to 1.8 the board appeared as a serial device after first flash and after setting the port in Arduino all subsequent flashes work like as the nano.

JiriBilek, 2020-11-26:
I hope it is not OT:
I had similar problems recently. The Blue Pill did not work on my new computer with external USB hub. After a couple of hours of investigation I realized I had forgot to fix the R10 issue on the board (change 10k resistor to 1k5). After fixing it uploading run smoothly.
It is really weird that the first time with the fresh bootloader, the USB enumeration succeeded but after flashing a first sketch it stopped to be seen. Therefore I wasn't suspicious about the resistor at first.