Serasidis/STM32_HID_Bootloader #31: Arduino IDE Settings for HID Bootloader, Bootloader Not Triggering

jalius, 2020-04-08:
Hi all! I've installed version 2.2.1 of the bootloader to my BluePill F103C8, but I'm having trouble working with it in the IDE. The hid-flash tool (visible from IDE window) only allows upload to my BluePill F103C8 when I have BOOT 1 in position 1 (BOOT 0 in position 0); I cannot upload otherwise. I assume the software is supposed to put it into bootloader mode automatically, but as it currently stands, my process to upload code is:

1. Unplug device, switch BOOT 1 to position 1.
2. Plug in device, hit upload on arduino software.
3. Wait for successful upload, then switch BOOT 1 to position 0, then manually trigger reset.

I currently mimic the software upload process, and it works, but I'd like to have it working more conveniently. Is there any detail about the specific Arduino tools menu settings I need to use?

[image]

fpistm, 2020-04-08:
To have to software trick to be able to switch in the HID bootloader mode you have to select the correct one.
In your case it seems you select COM8, is it the correct  port com  name for the Serial USB ?

jalius, 2020-04-08:
That was the issue! Wrong COM port! I think a computer restart was in order, as the COM port wasn't showing up at all before I restarted. Thank you, my friend.

jalius, 2020-04-08:
My friend, I'm getting the same issue again when I use my linux machine. It seems like there is an issue putting the board into a broken state. Here is my dmesg output:

> [  358.385981] usb 3-2: USB disconnect, device number 3
> [  480.593928] usb 3-2: new full-speed USB device number 4 using xhci_hcd
> [  480.743762] usb 3-2: New USB device found, idVendor=0483, idProduct=5740, bcdDevice= 2.00
> [  480.743768] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [  480.743772] usb 3-2: Product: BLUEPILL_F103C8 CDC in FS Mode
> [  480.743776] usb 3-2: Manufacturer: STMicroelectronics
> [  480.743778] usb 3-2: SerialNumber: 496D406A3734
> [  480.745187] cdc_acm 3-2:1.0: ttyACM0: USB ACM device
> [  496.540236] usb 3-2: USB disconnect, device number 4
> [  496.540478] cdc_acm 3-2:1.0: failed to set dtr/rts
> [  497.161578] usb 3-2: new full-speed USB device number 5 using xhci_hcd
> [  497.311465] usb 3-2: New USB device found, idVendor=1209, idProduct=beba, bcdDevice= 3.00
> [  497.311472] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
> [  497.311475] usb 3-2: Product: STM32F HID Bootloader
> [  497.311479] usb 3-2: Manufacturer: www.serasidis.gr
> [  497.341304] hidraw: raw HID events driver (C) Jiri Kosina
> [  497.344535] usbcore: registered new interface driver usbhid
> [  497.344536] usbhid: USB HID core driver
> [  497.349794] hid-generic 0003:1209:BEBA.0001: hiddev0,hidraw0: USB HID v1.11 Device [www.serasidis.gr STM32F HID Bootloader] on usb-0000:00:14.0-2/input0
> [  498.752656] show_signal_msg: 14 callbacks suppressed
> [  498.752665] hid-flash[4171]: segfault at c ip 0000000000403c70 sp 00007ffeb4f9f9b0 error 4 in hid-flash[400000+7000]
> [  498.752690] Code: b6 00 0f b6 c0 89 45 f0 c7 45 ec 00 00 00 00 83 7d f0 00 75 11 48 83 45 d0 01 48 83 6d c8 01 c7 45 ec 01 00 00 00 48 8b 45 d8 <8b> 40 0c 85 c0 7f 6a 48 8b 45 c8 0f b7 c8 48 8b 45 d8 8b 40 14 0f
> [  540.841285] usb 3-2: USB disconnect, device number 5
> [  542.116536] usb 3-2: new full-speed USB device number 6 using xhci_hcd
> [  542.266434] usb 3-2: New USB device found, idVendor=0483, idProduct=5740, bcdDevice= 2.00
> [  542.266441] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [  542.266446] usb 3-2: Product: BLUEPILL_F103C8 CDC in FS Mode
> [  542.266449] usb 3-2: Manufacturer: STMicroelectronics
> [  542.266452] usb 3-2: SerialNumber: 496D406A3734
> [  542.267931] cdc_acm 3-2:1.0: ttyACM0: USB ACM device

As we see, device connects OK and I can see the port ttyACM0 in Arduino software. Then, I run compile and upload:

> [  567.115496] usb 3-2: USB disconnect, device number 6
> [  567.117361] cdc_acm 3-2:1.0: failed to set dtr/rts
> [  567.739756] usb 3-2: new full-speed USB device number 7 using xhci_hcd
> [  567.889418] usb 3-2: New USB device found, idVendor=1209, idProduct=beba, bcdDevice= 3.00
> [  567.889425] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
> [  567.889428] usb 3-2: Product: STM32F HID Bootloader
> [  567.889431] usb 3-2: Manufacturer: www.serasidis.gr
> [  567.894132] hid-generic 0003:1209:BEBA.0002: hiddev0,hidraw0: USB HID v1.11 Device [www.serasidis.gr STM32F HID Bootloader] on usb-0000:00:14.0-2/input0
> [  569.328602] hid-flash[4573]: segfault at c ip 0000000000403c70 sp 00007ffed207ebf0 error 4 in hid-flash[400000+7000]
> [  569.328633] Code: b6 00 0f b6 c0 89 45 f0 c7 45 ec 00 00 00 00 83 7d f0 00 75 11 48 83 45 d0 01 48 83 6d c8 01 c7 45 ec 01 00 00 00 48 8b 45 d8 <8b> 40 0c 85 c0 7f 6a 48 8b 45 c8 0f b7 c8 48 8b 45 d8 8b 40 14 0f

[image]

When I plug it back in, it resets and becomes visible again in software. By

fpistm, 2020-04-08:
About the  hid_flash segfault, it seems an known issue and PR's have been provided to solve this:
#30 and #15.

TheKikGen, 2020-04-29:
Also, don't forget the DVECTOR command line parameter : your code must be compiled accordingly with the size (and version) of the hid bootloader your are using and your linkage file must be adjusted.
If bootloader size > 2K : DVECTOR = 0X8001000
below 2K  DVECTOR = 0X8000800 

Also, know that high density devices are not well supported...