Serasidis/STM32_HID_Bootloader #34: stm32 shows up on dmesg but not in arduino ide

Abh15h3k, 2020-05-12:
STM32 Board: STM32F103C8T6 Blue Pill, PC13 led.
OS: Archlinux x64

After setting boot0 to 1.
I uploaded the F103/low_and_medium_density/hid_generic_pc13.bin bootloader to my blue pill using stm32flash (sudo stm32flash -g 0x8000000 -b 115200 -w hid_generic_pc13.bin /dev/ttyUSB0). Then before disconnecting the ftdi programmer i moved boot 0 back to 0.

But the stm32 does not show up in arduino.

But dmesg does show that the stm32 is connected to the pc.

[image]

What is the issue?
as it is i am a bit confused with the stm32 board. I am sorry if i am doing something wrong or stupid.

roboticboyer, 2020-05-20:
Under Linux you need to update the [udev roles](https://github.com/Serasidis/STM32_HID_Bootloader#linux-udev-setup)

albertskog, 2020-05-24:
Hi! I'm also running a Bluepill with STM32F103C8 but under Ubuntu 20.04 and seeing the same problem..

I have installed the udev rules:
```bash
➜  ~ cat /etc/udev/rules.d/99-stm32_hid_bl.rules
# STM32_HID_bootloader
ATTRS{idProduct}=="beba", ATTRS{idVendor}=="1209", MODE:="666"
➜  ~ sudo udevadm control --reload-rules
```

And after flashing the bootloader, it shows up in `dmesg` but I still can't "find" it in `/dev/`...

```bash
➜  ~ dmesg | tail
[12859.015618] usb 1-1.1.4.2: new full-speed USB device number 88 using ehci-pci
[12859.125231] usb 1-1.1.4.2: New USB device found, idVendor=1209, idProduct=beba, bcdDevice= 3.00
[12859.125239] usb 1-1.1.4.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[12859.125244] usb 1-1.1.4.2: Product: STM32F HID Bootloader
[12859.125248] usb 1-1.1.4.2: Manufacturer: www.serasidis.gr
[12859.127117] hid-generic 0003:1209:BEBA.0021: hiddev4,hidraw5: USB HID v1.11 Device [www.serasidis.gr STM32F HI
D Bootloader] on usb-0000:00:12.2-1.1.4.2/input0
```

I'm trying to run `hid-flash` manually with `any` as the comport and it seems to partially work:

```bash
➜  ~ ./hid-flash firmware.bin any

+-----------------------------------------------------------------------+
|         HID-Flash v2.2.1 - STM32 HID Bootloader Flash Tool            |
|     (c)      2018 - Bruno Freitas       http://www.brunofreitas.com   |
|     (c) 2018-2019 - Vassilis Serasidis  https://www.serasidis.gr      |
|   Customized for STM32duino ecosystem   https://www.stm32duino.com    |
+-----------------------------------------------------------------------+

> Trying to open the [any]...
error, counldn't open [/dev/any]
> Unable to open the [any]
> Searching for [1209:BEBA] device...
#
> [1209:BEBA] device is found !
> Sending <reset pages> command...
> Flashing firmware...
. 1024 Bytes
. 2048 Bytes
. 3072 Bytes
. 4096 Bytes
. 5120 Bytes
. 6144 Bytes
. 7168 Bytes
. 8192 Bytes
. 9216 Bytes
. 10240 Bytes

> Done!
> Sending <reboot mcu> command...
> Searching for [any] ...
error, counldn't open [/dev/any]
error, counldn't open [/dev/any]
error, counldn't open [/dev/any]
error, counldn't open [/dev/any]
error, counldn't open [/dev/any]
> Finish
```

After this however, my blinky firmware does not run correctly, the led is just constantly on and I have to re-plug the board with BOOT**1** set to 1 to get back to the bootloader so I can try again.

Any ideas?

Abh15h3k, 2020-05-24:
I am having the same issue as @albertskog.
After flashing the "hid_generic_pc13.bin" using stm32flasher and a FTDI serial flasher the green led flashed violently. i move boot 0 back to '0'. connect via onboard micro usb. does not show up in arduino ( with the udev rule or when run with sudo ). i try to flash directly with hid-flash. it says flashing firmware .... done. and the green led is always on.

[image]

roboticboyer, 2020-05-24:
I have collected my notes on [this page](https://hardwareliberopinerolo.github.io/Pages/BluePill)

The USB is mounted as `/dev/ttyACM0` or `/dev/ttyACM1` not `/dev/any`

Remember to use the low and medium density STM32_HID_Bootloader ( id_generic_pc13.bin )

albertskog, 2020-05-24:
Sorry, I think I was a bit unclear - thats exactly my problem, I don't have either of those. I just used "any" as a test to show that it still looks like the cli can talk to my device.

roboticboyer, 2020-05-24:
I've used the CLI version of STM32CubeProgrammer as done by Arduino IDE to uplaod the STM32_HID_Bootloader

`arduino_STM32/portable/packages/STM32/tools/STM32Tools/1.3.2/tools/linux/stm32CubeProg.sh 1 /tmp/hid_generic_pc13.bin ttyUSB0 -s`

albertskog, 2020-05-24:
Thanks @roboticboyer, I had not tried that before but now i have flashed the bootloader several times  using all of these methods:

* `st-flash` + STLink
* `stm32flash` + FTDI cable 
*  `stm32CubeProg.sh` + FTDI cable 

As you can see in my dmesg printout above, the bootloader seems to be flashed correctly to the board, otherwise dmesg should not print out  Product: STM32F HID Bootloader` and `Manufacturer: www.serasidis.gr` etc. I think my problem might be with the udev rules? I did triple-check everything, tried both the reload command and reboot, but still can't find a serial port..

roboticboyer, 2020-05-24:
As rules I used `97-stm32_hid_bl.rules` not `99-stm32_hid_bl.rules` because there was another rules as 99

roboticboyer, 2020-05-24:
see what there is with: `ls /etc/udev/rules.d/`
Rules 98 and 99 could be used by other Arduino boards

albertskog, 2020-05-24:
Thanks, I tried changing to `97-stm32_hid_bl.rules` and reloading rules again but there was still no new device in /dev/ after re-plugging the board.

I also checked my other file [99-platformio-udev.rules](https://raw.githubusercontent.com/platformio/platformio-core/master/scripts/99-platformio-udev.rules) for conflicts with the vendor and product id but and there weren't any. 

A little off-topic: As far as I can tell, the numbers are just to determine in which order rules are parsed so it should be fine if other files also start with 99. I had to double check to make sure and found this: https://unix.stackexchange.com/questions/204979/why-do-the-rules-in-udev-rules-d-have-numbers-in-front-of-them

Abh15h3k, 2020-05-24:
Ok. so thanks to @roboticboyer notes i found the video for flashing the stm32. i was able to successfully use the HID bootloader for 3 uploads. after that i wanted to enable HID mode for the stm32. i did that in arduino. then uploaded to the port `ttyACM0` which had succeeded but from this point on i was not getting any port or able to upload to it.
i even repeated the bootloader flashing multiple times. no luck.

right now i am 'ok' with flashing the stm32 via the ftdi programmer using stm32prog.
I will still attempt to get the HID bootloader working. 

I have the udev rule in /etc/udev/rules.d (there were no other rules).
flashed the bootloader following @roboticboyer notes through the terminal.

roboticboyer, 2020-05-24:
Remember to activate Serial in USB and USART option or you lose the USB port and you need to reupload the bootloader
See my working setup below:
[image]

Abh15h3k, 2020-05-24:
Ok. as long as i keep `USB support = CDC(generic 'Serial' supersede U(S)ART)` the port is there. but for my use case i need to change it HID. that causes me to loose the port. Is this the expected/intended behaviour?

albertskog, 2020-05-25:
Thanks for pointing that out @Abh15h3k! I just read the text but did not watch the video until now.
I checked and my board also works with those settings using Arduino on my other computer which is a mac.

I also confirmed that once it worked on my mac, the serialport also appears on mu Ubuntu machine. Now I can go back and try to get it working in Platformio on the Ubuntu machine, probably some configuration error on my side!

A note to anyone else having this problem in the future, and that I think answers your question @Abh15h3k:
The thing I did not realise until now is that the serialport that we are looking for is not "part of" the bootloader but it is the regular serialport that can be used in the sketch. It is just used to reset the board to get back to the bootloader. This is why you can not use it with HID mode in the sketch and it's probably why mine is not showing up at all (because I did not enable serial properly).

Abh15h3k, 2020-05-26:
Yeah. that seems to be the case.
Thanks @roboticboyer for helping solve this.

closed 2020-05-26 by Abh15h3k