Serasidis/STM32_HID_Bootloader #36: hid-flash segmentation fault

pabolojo, 2020-07-04:
I am trying to flash my STM32F103C8, I have already compiled my blink test sketch but when I try to flash the .bin I get a segmentation fault.

I execute it with this command:
```
./hid-flash ~/Desktop/test.bin ttyACM0
```

The output that I get is:
```
+-----------------------------------------------------------------------+
|         HID-Flash v2.2.1 - STM32 HID Bootloader Flash Tool            |
|     (c)      2018 - Bruno Freitas       http://www.brunofreitas.com   |
|     (c) 2018-2019 - Vassilis Serasidis  https://www.serasidis.gr      |
|   Customized for STM32duino ecosystem   https://www.stm32duino.com    |
+-----------------------------------------------------------------------+

> Trying to open the [ttyACM0]...
error, counldn't open [/dev/ttyACM0]
> Unable to open the [ttyACM0]
> Searching for [1209:BEBA] device...
#
> [1209:BEBA] device is found !
> Sending <reset pages> command...
Segmentation fault
```

After that the STM32's led starts to blink quickly, and I have to restart the STM32 to keep executing the sketch that is flashed in it.

I am using a Raspberry Pi 4 as host system.

pabolojo, 2020-07-04:
After installing the udev rule it seems to work well!

closed 2020-07-04 by pabolojo

jerkey, 2023-07-21:
this isssue is affected by this PR which hasn't been merged yet for some reason
https://github.com/Serasidis/STM32_HID_Bootloader/pull/30

basically the segmentation fault gives no info, but with this fix, hid-flash will correctly inform you "Unable to open the [0x1209:0xBEBA] device.