Serasidis/STM32_HID_Bootloader #38: Maple Mini board not connected?

dbhaig, 2020-07-28:
Hello.
I am running Ubuntu Linux

I have loaded the hid_maple_mini.bin onto my maple mini clone board, following the procedure describe here: https://www.youtube.com/watch?v=Myon8H111PQ

When I plug in the board,
  
lsusb returns: ID 1209:beba InterBiometrics

ls /dev/ttyUSB*  returns nothing

When I try to upload a sketch using the Arduino IDE I get the following messages:

error, counldn't open [/dev/ttyUSB0]

+-----------------------------------------------------------------------+
|         HID-Flash v2.2.1 - STM32 HID Bootloader Flash Tool            |
|     (c)      2018 - Bruno Freitas       http://www.brunofreitas.com   |
|     (c) 2018-2019 - Vassilis Serasidis  https://www.serasidis.gr      |
|   Customized for STM32duino ecosystem   https://www.stm32duino.com    |
+-----------------------------------------------------------------------+

> Trying to open the [ttyUSB0]...
> Unable to open the [ttyUSB0]
> Searching for [1209:BEBA] device...
#
> [1209:BEBA] device is found !
> Sending <reset pages> command...
the selected serial port > Sending <reset pages> command...
 does not exist or your board is not connected

What am I missing?
Thanks

Riffer, 2020-08-08:
Which version of STM32 core did you install?
I had a negative experience with 1.9 and 2.2.2. maybe you give 1.8 a try.

dbhaig, 2020-08-09:
I am using 1.9.0. I will give 1.8.0 a try.

Thanks