Serasidis/STM32_HID_Bootloader #40: Blue pills code looping on procedure return

rromano001, 2020-09-27:
Hello, I am not an Arduino user and lack of debugger is a terrific issue.
 I loaded a test program from web,
https://maker.pro/arduino/projects/simple-fm-radio-receiver-with-arduino-uno-and-rda5807m
 loaded doesn't work as is. 
Adapting got some time to isolate where is stuck. Code is quite noobs and seems modified from other sources but issue come from library or bootloader.
Need a lot of wait sparse time before use serial communication.
 At final code stuck at wire communication, procedure never return and loop as stack got corrupted, reenter procedure?
 No debugger no idea what happen... Arduino IDE is just a time wasting.
changed key selector from analog to digital, added timeout on exit.
stuck on status read where never return back to main loop.

init ok
sensor = 98
Post butt
Millis=2963 Time=2912 Time1=2912 Time2=2912 Time3=2912
Pre StatRDReqfrom for Loop after for End proc readST
Reqfrom for Loop after for End proc readST
Reqfrom for Loop after for End proc readST
Reqfrom for Loop after for End proc readST
Reqfrom for Loop after for End proc readST
Reqfrom for Loop after for End proc readST
Reqfrom for Loop after for End proc readST
Reqfrom for Loop after for End proc readST

2 3 second hang, stack overlow?
Reqfrom for Loop after for End proc readST
Reqfrom for Loop after for End proc readST
Reqfrom 

[rdatest.ino.zip](https://github.com/Serasidis/STM32_HID_Bootloader/files/5288596/rdatest.ino.zip)

got similar issue from rf generator test file, this is <SPI out only> no hardware required for test.

[ADF4351.zip](https://github.com/Serasidis/STM32_HID_Bootloader/files/5288593/ADF4351.zip)

rromano001, 2020-09-28:
Found the possible source of original radio library from maker.pro:
http://www.mathertel.de/Arduino/RadioLibrary.aspx
or from groove library:
https://wiki.seeedstudio.com/Grove-I2C_FM_Receiver/

 Examples require again add delay on setup serial library.
 Mathertel:  serial radio work, Lcdradio not but require adaptation. Serialscan loop on scan but never stop as if wire was broken. 
 Grove too doesn't work, loop on seek without stopping.
 Radio hardware is Ok tested with serialradio example.

 Next step is to migrate to mxCubeIDE code, this platform is unreliable in my point of view.
Mathertel code look professional grade but other are just looking  beginner level hash of mathertel  lib or grabbed somewhere.

[image]

rromano001, 2020-09-28:
Here file adapted from examples to just not crash on BluePills.
[fmradio.zip](https://github.com/Serasidis/STM32_HID_Bootloader/files/5291440/fmradio.zip)

KenjutsuGH, 2020-09-30:
Just to understand better, are there any problems with the **STM32_HID_Bootloader** then?

rromano001, 2020-09-30:
Hi Kenjutsu, yes I am experiencing trouble with serial 
communication, STM32 code hang.
  I found workaround of inserting delay but doesn't solve issue.
  Attached Salea LogicPro8 to see what happen USB subsystem freezed 
up. Removed Salea and all other peripheral, on Linux mint 20 HID 
serial communication hang and reboot is required to restore USB 
functionality.
  As I wrote not an Arduino fan so no idea how to debug code.
  Thank you.
  Roberto

rromano, romebay both are alias of same email

KenjutsuGH, 2020-09-30:
What options do you have under **Tools**?

rromano001, 2020-09-30:
Hi, s screenshot of tools,
  Board generic stm32f103
Blue pill f103c8
usart generic serial
usb cdc generic supersedes
usb low/full speed
optimize smallest -os (default)
runtime newlib nano (default)
HID Bootloader 2.2
Regards
  Roberto

rromano, romebay both are alias of same email

KenjutsuGH, 2020-10-01:
Hi,

I am using macOS 10.13.6 so I can not help you with the Linux stuff. I have the same settings under **Tools**. I also noticed that as soon as **Serial.begin(115200)** is executed, it takes about 1.5s before I would see output in the serial monitor. 

As an example, I modified the classic **Blink** sketch to output some info to the serial port:
```
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
  Serial.begin(115200);  
  Serial.println("Serial setup done");
}

// the loop function runs over and over again forever
void loop() {
  Serial.println("LED_BUILTIN, HIGH");
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  Serial.println("LED_BUILTIN, LOW");
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}
```

After the upload, I never see the "Serial setup done" message being displayed. Instead of using a **delay()** statement after **Serial.begin()** , I use

`while (!Serial);`

With the above code, all the info is displayed in the Serial Monitor and the program keeps on running. I can close the Serial Monitor and use a program such as **CoolTerm** (also available for Linux) to connect to the serial port of the BP and the output continues to be displayed.

Unfortunately, I do not have a RDA5807M module to test your program. I do have the same I2C OLED display, and when I upload your program, I see info on the OLED, and the following in the Serial Monitor:
```
init ok
sensor = 0
Post butt
Millis=2975 Time=2925 Time1=2925 Time2=2925 Time3=2925
Post millisFrecuencia = 210
Post FreqPost Volume>BLOCK_E found.
End Loop
sensor = 0
Post butt
Millis=3147 Time=2925 Time1=2925 Time2=2925 Time3=2925
Pre StatRDReqfrom for Loop after for 
```

fpistm, 2020-10-01:
https://www.arduino.cc/reference/en/language/functions/communication/serial/ifserial/

rromano001, 2020-10-01:
Hello, thank for prompt answer, I suppose on MAC it work hassle 
free. On Linux too many time it lock out but this is strange, so 
inspect code is more difficult. I suspect also some Linux incoerence, 
as if some wrong code injected. Maybe M$ is poisoning code?
  About RDA module can send you one for free but from your writing 
sketch has no issue on mac.
  Regards
  Roberto

rromano, romebay both are alias of same email

KenjutsuGH, 2020-10-01:
Have you tried another upload method such as SWD or Serial?