Serasidis/STM32_HID_Bootloader #44: HID bootloader and STM32F103R

GitMauri, 2021-01-27:
Hi Serasidis

I'm using Roger Core

I've used your Bootloader on Roger Clark's core and it was working great on 103c8t6... now I've tried to use on STM32F103RCT6 and there are problems that I cannot fix... (STM32 HID Bootloader 2.1.0)

I tried to add an edited hid_bootloader file to ....\STM32F1\variants\generic_stm32f103r\ld and I edited the boards.txt file for genericSTM32F103R UPLOAD METHODS adding HID....

I tried to recompile the generic-PD2.bin (make generic-pd2 PAGE_SIZE=2048) ...

When I upload the sketch the IDE recognizes the board and it seems to upload correctly, but a simple blink doesn't work...
(It works with Roger bootloader).

Maybe the HID bootloader is just not compatible with high density devices..?

Can I use the board as a 128KB STM32F103RBT6? That should be a low density right?

I tried but without luck... I compiled the generic-PD2.bin on Windows using CYGWIN64 and the GNU Arm Embedded Toolchain: 10-2020-q4-major December 11, 2020. It compiles without errors but it gives some warning about deprecated things...

the board: https://stm32-base.org/boards/STM32F103 ... -V2.0.html
The chip onboard seems to be genuine... in the past I had issues with blue pills that worked ok with Roger bootloader, but they all couldn't work with the HID bootloader... turned out those blue and black pills had on board all fake chips...

I would like to keep Roger Core because I'm working on a project using arprus USB_composite (which is only compatible with Roger core)

Thanks

Maurizio

TheKikGen, 2021-01-27:
check that issue already posted here : https://github.com/Serasidis/STM32_HID_Bootloader/issues/32

TheKikGen, 2021-01-27:
or try this one : https://github.com/TheKikGen/stm32-tkg-hid-bootloader

GitMauri, 2021-01-28:
> or try this one : https://github.com/TheKikGen/stm32-tkg-hid-bootloader

Hi Kik

Before to open the issue I've already checked your suggestions and your repo, of which I'm one of the followers.
I'm happy to see that the HID bootloader has got fresh improvements and new features from you, it is a great new by itself bootloader, and I hope I will be able to use it soon.

Here the results of my tests.

As you know Serasidis bootloader started with Roger Core. but stopped the development for Roger Core to version 2.1
2.2.1 and 2.2.2 are optimized for the official STM32 Arduino core, and not compatible with Roger's core.

**I'm using Roger Clark's Core and IDE 1.8.5**

I tried your fix to the file HID.c with a lot of hope, but aware it could have been incompatible with Roger's core.
After changing the HID.c with you part of code for static void HIDUSB_HandleData(uint8_t *data) I tried to recompile generic_D2.bin, but could not compile.... many variable missing... so I guessed you modified the last 2.2.2 bootloader file....

Then I thought: this kik seems to have looked inside the codes a fairly amount of time... let's check his Repo....  then i found your bootloader, and obviously I tried immediately.... 
First test : 
I injected your generic_D2.bin by st_link, and i've read your bootloader kept the compability with the  original HID-FLASH.
So i gave it a try....
Same result as for Serasidis bootloader: Board recognized, start the upload... but then the sketch doesn't work.

Second test:
I installed your  TKG-FLASH with the various add to the board file and adding an hid_bootloader_tkg_r.ld file

```
genericSTM32F103R.menu.upload_method.HIDUploadMethod2K=TKG-HID bootloader
genericSTM32F103R.menu.upload_method.HIDUploadMethod2K.upload.tool=tkg-hid
genericSTM32F103R.menu.upload_method.HIDUploadMethod2K.build.upload_flags=-DSERIAL_USB -DGENERIC_BOOTLOADER
genericSTM32F103R.menu.upload_method.HIDUploadMethod2K.build.vect=VECT_TAB_ADDR=0x8001000
genericSTM32F103R.menu.upload_method.HIDUploadMethod2K.build.ldscript=ld/hid_bootloader_tkg_r.ld

```

inside the folder .....STM32F1\variants\generic_stm32f103r\  hid_bootloader_tkg_r.ld 

```
MEMORY
{
  ram (rwx) : ORIGIN = 0x20000000, LENGTH = 48K
  rom (rx)  : ORIGIN = 0x08001000, LENGTH = 248K
}

/* Provide memory region aliases for common.inc */
REGION_ALIAS("REGION_TEXT", rom);
REGION_ALIAS("REGION_DATA", ram);
REGION_ALIAS("REGION_BSS", ram);
REGION_ALIAS("REGION_RODATA", rom);

/* Let common.inc handle the real work. */
INCLUDE common.inc
```

added your TKG-FLASH  to the win folder... 

Then I started to compile...  and.....

.....  almost there......  yes, maybe ..... and ....

**BOOOOMM**

**error while uploading: missing 'upload.parms.quiet' configuration parameter**

So I thought maybe your bootloader is not compatible with Roger's Core? I've tuned many small things that are not compatible with the new STM official core, so I'd like to use Roger's Core...  I thought to open an issue on your Repo, but you never declared the compatibility with Roger's core, so if it is not designed to be compatible maybe I should pull a request? 

Or maybe it is compatible but I'm missing some steps?

Thanks

Maurizio

TheKikGen, 2021-01-28:
Hi Maurizio.  

- My project lost compatibilty since the very recent v 3.1. I have to correct that point in the wiki.
- I also use the Roger's stm32duino core with my own projects.  No compatibility issues at all.
- The error your are mentioning is linked with the IDE integration.  Try to to upload out of the IDE first :

1/ Set the upload method to the TKG HID.  This will change the  rom origin to 0x08002000 for you.
2/ Hit Ctrl + Alt +S to export binaries : you will get the bin file compiled by the IDE in your sketch directory. 
3/ Copy the TKG-FLASH utility in your project directory
4/ Set the board to bootloader mode with a  "double click" on the reset button. The LED should flash rapidly
5/ Open a command line window, and enter "tkg-flash -info" => You should see some info from the bootloader about the uC
6/ Flash your firmware : "tkg-flash <your bin file>"

I will check the IDE integration later....

note : When embedded in the IDE, If you are using anything else than a serial usb in your projet (e.g. like me, usb midi), you can't use the DTR reset method to reboot the board, so you must reboot manually by pressing the reset button at the right time.

GitMauri, 2021-01-28:
Hi Kik

I'm a little confused, you mean I don't have to try with the last 3.1 but with the previous?

I tried with the 3.1 but no luck...

I changed the ld file as suggested rom (rx) to 0x08001000.

I compiled a binary with Board 103R generic 48 / 256...  and as bootloader your TKG-HID...
Then I tried flashing by your tkg-flash...

Board recognized, started the upload procedure but it stopped with an error...

```

+-----------------------------------------------------------------------+
|             TKG-Flash v3.1 STM32F103 HID Bootloader Flash Tool        |
|                     High density device support.                      |
|       (c) 2020 - The KikGen Labs     https://github.com/TheKikGen     |
+-----------------------------------------------------------------------+
Build : 3.210120.1703

> Searching for [1209:BEBA] HID device...|
> [1209:BEBA] HID device found !
> Firmware file size is 16212 bytes.
> Flashing firmware :
  INFO - Informations reported by the bootloader :
      Firmware version       : 0310
      MCU Flash memory size  : 256 K (high density device)
      Page size              : 2048 bytes
      Page offset            : 2 pages
      Flash base address     : 0x08001000
  START command sent successfully.

  [..................................................]
  ** Error while sending firmware data to the HID device.
> TKG-Flash end.
```
In the while maybe I try the 2.4.1

Thanks for your help

GitMauri, 2021-01-28:
Hi

Also with the 2.4.1 no luck... I chosen the generic PD2 and I tried with the old and new tkg-flash... 
 The strange thing is if I choose Serasidis HID I can see the Ide pushing all the 16k but the Sketch doesn't work... If i use Roger bootloader it goes and works.. 

I start to think maybe there is a problem with my MCU? Could be a fake one... but it looks genuine. In the past I had blue and black pills that were working with Roger bootloader but not with the HID bootloader...  A close look to the chip with a good magnifier and it was evident that they were fake (and very very cheap).

But if the chip is genuine, maybe is a problem with the tkg-flash file?

GitMauri, 2021-01-28:
Here the development: think the problem is your tkg-flash, maybe small small bugs
I remembered that i didn't fix the ld file and the board file with the new value for Serasidis.... so I tried to do it again...
I uploaded the Serasidis bootloader bin file and tried with his hid-flash... but nothing changed...

Then I thought that the bin file was recompiled by me with new toolchain, and i thought maybe it could have had some problems...  So i tried again with your bootloader bin file..... and again with the HID-flash from Serasidis with this time the corrected ld and board values... and....

and..... maybe .....

**YES.......   YES YESSSSSSSS...**

It worked..... the board reenumerated, and was recognized as a maple on com port and the led blinked ... blinked so well :-)
double reset and tried again... in HID... perfect.... tried on maple com port, again OK

So we know your bootloader works, but there is something to fix on your tkg-flash

Thanks so much!   Let me know how can i help to debug your tkg-flash

Now I go to sleep here it is 23:20 of Thursday

Maurizio

TheKikGen, 2021-01-29:
Hi Maurizio.  I wasn't able to reproduce the issue under Linux, but I tested more deeply under windows 10, and in some cases, the upload stucks.  It is because windows is by far, much slower to enumerate and setting everything around usb.  I added a timing right after opening the HID device, and all seems ok now.  Are you using windows platform ?

So :

1/ Flash again the tkg hid bootloader v3.10 at 0X8000000

2/ Use the attached  tkg-flash.exe I built for testing purpose, and flash your "blink" sketch
I have amended the  documentation https://github.com/TheKikGen/stm32-tkg-hid-bootloader.  Follow the section "Arduino IDE  integration" to get a clean IDE with that booloader.  Do not mix 2 distributions....

If you still have an issue, please open it on the issue tab here : https://github.com/TheKikGen/stm32-tkg-hid-bootloader/issues 

[tkg-flash.zip](https://github.com/Serasidis/STM32_HID_Bootloader/files/5891125/tkg-flash.zip)

GitMauri, 2021-01-29:
Hi Kik

Thanks for your work! If you are in France it should be very late right now

But I have 2 news, one good and one will be an issue in your Repo.

When you asked me if I was using Windows 10 and that you had the problems just sometime a small bulb lighted in my brain...
The answer is no, I use Arduino on W7 64 professional machines , 2 actually with 2 different "releases" of Roger... one requires the HID patch from Serasidis and one doesn't.... So this already pointed me toward the possible problem... WINDOWS 7 compatibility... however I gave a try on both W7 machines with the sketch compiled in 2 different machines... both failed...

Then I have a Windows 10 machine that I use for Graphic Design, video editing and other staff that requires more steam...

Took the board, I uploaded on the desktop a folder with the compiled sketch lunched the exe you just gave me for test ...

and.......... maybe.......

```

+-----------------------------------------------------------------------+
|             TKG-Flash v3.1 STM32F103 HID Bootloader Flash Tool        |
|                     High density device support.                      |
|       (c) 2020 - The KikGen Labs     https://github.com/TheKikGen     |
+-----------------------------------------------------------------------+
Build : 3.210129.0246

> Searching for [1209:BEBA] HID device...|
> Searching for [1209:BEBA] HID device...-
> [1209:BEBA] HID device found !
> Firmware file size is 16196 bytes.
> Flashing firmware :
  INFO - Informations reported by the bootloader :
      Firmware version       : 0310
      MCU Flash memory size  : 256 K (high density device)
      Page size              : 2048 bytes
      Page offset            : 2 pages
      Flash base address     : 0x08001000
  START command sent successfully.

  [++++++++++++++++++++++++++++++++++++++++++++++++++]   16384 bytes sent.
  No checksum error.

> Firmware flashing done !
  16 sectors written.
> TKG-Flash end.
```

Yes, it works on Windows 10.... but for some reason it doesn't work on Windows 7
It would be nice to have it working on Windows 7 too, I'm going to open an issue

Thanks so much

Maurizio

TheKikGen, 2021-03-07:
If you reach this point : discussion continue here  https://github.com/TheKikGen/stm32-tkg-hid-bootloader/issues/6