Serasidis/STM32_HID_Bootloader #47: STM32F407G-DISC1 compatibility with HID Bootloader

HappySamuel, 2021-07-16:
Hi

As i saw there is stm32f407 bootloader support, so I bought a STM32F407G-DISC1 board. However, when i successfully uploaded the HID bootloader, the direct USB (known as STM32 Virtual COM Port) doesn't appear in Device Manager on Windows Neither can I find it on Arduino IDE.

I guess i must have missed some steps. Please guide me the steps on getting it up to work. Or is there any video link that show the steps like the one for bluepill board using HID bootloader?

Best,
Samuel

geraldjust, 2021-08-20:
you need to upload the bootloader via SWD. try using stm32cube Programmer (programming USB port). To enter bootloader mode make sure you press the button you designated for it as your plugging in power/USB cable. Windows should appear as HID device. ( using native USB port)

under arduino select the right board. make sure usb CDC is on. Serial set to generic serial. Upload method as HID 2.2 . Upload any sketch to it. From here on out, as long as you got usb CDC on the sketch will auto upload via HID without the need of the "plugin while pressing button" way.

closed 2021-08-25 by HappySamuel

Pey-mma, 2023-12-24:
Hello everyone,
I just wanted to thank you all for the helps you did to me in this discussion.
And if I could ask, I have a problem that after flashing the bootloader onto the STM32F407G-disc1, I couldn't program using the HID bootloader also the serial monitor doesn't work neither.
Any idea of what could be wrong? 
Appreciate that