Serasidis/STM32_HID_Bootloader #48: How can I change MAX_PACKET_SIZE to 64?

myj-dev, 2021-07-29:
When I change MAX_PACKET_SIZE to 64, the PC will fail to request device descriptor.

geraldjust, 2021-07-29:
did you change the MAX_PACKET_SIZE on the CLI or bootloader?

myj-dev, 2021-07-29:
> did you change the MAX_PACKET_SIZE on the CLI or bootloader?

I changed the MAX_PACKET_SIZE macro definition in bootloader. Then PC can not enumerate the device.

geraldjust, 2021-07-29:
I think you need It to be 65. As the first byte is used as the report ID. Which is 0 but it's still needed for proper communication.

myj-dev, 2021-07-29:
> I think you need It to be 65. As the first byte is used as the report ID. Which is 0 but it's still needed for proper communication.

Thanks for your advice. I tried setting MAX_PACKET_SIZE to 0x10, 0x20, 0x40, 0x41. Only value 0x10, 0x20 and the original value 0x08 can work properly.

geraldjust, 2021-07-29:
Oh your trying to modify the F1 bootloader. I think that's part of the optimization in size for the F1. It need to cycle though the 8 bytes 8 times to fill the 64 byte buffer that would be on the F4. Then the bootloader needs it to be 8 I think. Sorry my original answer was inferring you were using the F4.

closed 2021-08-02 by myj-dev