Serasidis/STM32_HID_Bootloader #49: USB Com not available

geocheats2, 2021-08-19:
Hello to everyone

I have a custom PCB with STMf407IGT and i have ported Arduino_Core_STM32 and works just fine when flashed using ST-LINK
and the serial com works, i compiled STM32_HID_Bootloader for my MCU (diff clock) and while it flashes just fine using the following settings on Arduino IDE i don't seem to be able to make the USB Com Port available on my device manager.

Do you have any ideas at what to look??

George Kalampakas

My settings

[image]

strannyi, 2023-01-10:
Hi, i see same issue. My board is bluepill and i use STM32_HID_Bootloader-2.2.2, Arduino IDE 2.0.3.
The Windows 10 system found device, but no Virtual COM port was created. Need help.

[image]

FabioM3, 2023-12-20:
I think you don't have to select a COM port.

gitarmpit, 2024-05-02:
> I think you don't have to select a COM port. And arduino ide 1.85+ don't work with hid-bootloader. But you have a lower version.

1.85?  What is that?   Arduino versions are three digits, like maybe 1.8.5 but that's too old, the newest 1.8.x I could find is 1.8.19 from 2021.  Version 2 became available around that time.