Serasidis/STM32_HID_Bootloader #51: USB device not recognized

dmkng, 2022-01-16:
Hello,
I flashed the hid_generic_pc13.bin file to my BluePill board using a serial dongle and STM32CubeProgrammer with the default start address (0x08000000) and also the start address specified in README (0x8000000). When plugging in the board via USB the LED doesn't blink or anything, and after 1 second I'm getting a Windows notification "USB device not recognized". Of course I can't flash anything using Arduino IDE because hid-flash can't find any HID device.
It is not the board's fault because STM32duino bootloader works perfectly, LED flashes when plugged in after flashing via serial dongle and I can successfully flash sketches via USB. What can be the problem? Please help. Thanks.

bance, 2022-02-13:
same fault here with 2.2.1 and 2.2.2

bootloader flashed with st-link v2. verified ok.

bance, 2022-04-03:
thank you for the solution, dmkng!

Istria1704, 2022-06-24:
Thank you!
I had the identical problem. After a couple of hours of searching around and experimenting, it turned out to be this. 

Can conform the @dmkng fork is the fix for STM32F103C6T6 boards.

Thanks again!

closed 2023-01-17 by Serasidis