Serasidis/STM32_HID_Bootloader #55: New Project for STM32F4x1 Bootloader

xromansx, 2022-12-06:
Hi, 
 I tried to compile one from here even with modifications I couln't get it working on STM32F401 so decided to port it to PlatformIO for easie compiling. I can build now the bootloader it detects as HID and Led lights up. but I can't make it to work with hid-flash. If someone can help me to troubleshoot it or to point in corect direction we may have good and reliable bootloader for 4x1 series.
ling to my git [](https://github.com/xromansx/STM32F4x1_HID_Bootloader )

arthurfprecht, 2024-08-08:
Hi @xromansx, I had a similar issue, but after some time I made it to work, I will try to compare the src folder in your repo with what I have and then make a pull request 😄

xromansx, 2024-08-21:
This would be nice, Thank you.