Serasidis/STM32_HID_Bootloader #60: Device support issue for Bluepill stm32F10336t6

TheSnowFall, 2023-04-11:
Hello,
My blue pill contains a chip of stm32F10336t6.
I am using STM32 core by STMicroelectronics (Arduino IDE).
I followed all the processes of installing this HID bootloader. But when I connect the USB cable to my Windows PC it says,

`unknown USB device (device descriptor request failed)`

How I can solve this?

mcer12, 2023-08-09:
You probably mean STM32F103C6T6, I also have these chips (from aliexpress) and I wasn't able to make this bootloader work. STM32F103C8T6 works fine. Mind that C6T6 is a different - lower tier - part, it doesn't differ only in memory but it lacks some peripherals. They for example lack second SPI interface. That may be the reason why this bootloader doesn't work.

AgainPsychoX, 2024-08-29:
I have the same issue I think.

dmkng, 2025-03-23:
Hello, I've had the same problem long time ago (#51). Typical Blue Pill board uses STM32F103C6T6 (32MB RAM, 10MB Flash) but this bootloader supports only STM32F103C8T6 (64MB RAM, 20MB Flash), I created a fork where I roughly fixed this issue creating new target "bluepill". Here's my fork: https://github.com/dmkng/STM32_HID_Bootloader. Note that the HID bootloader only works with stm32duino/Arduino_Core_STM32, not with rogerclarkmelbourne/Arduino_STM32, but I will try to fix it in my fork soon.