Serasidis/STM32_HID_Bootloader #62: Windows not recognising the device

vibhoregit, 2023-06-22:
After uploading the boot loader and connecting the blue-pill with USB cable, the board doesn't show up as a COM port or HID device but unknown USB device. The resistor on board are correct and older generic serial boot loader seems to work just fine.    
Windows description:

Windows has stopped this device because it has reported problems. (Code 43)
A request for the USB device descriptor failed.

arthurfprecht, 2024-08-08:
I am also experiencing this. A similar issue is reported in https://github.com/Serasidis/STM32_HID_Bootloader/issues/64

Harry1326, 2024-08-15:
please refer to this:  https://blog.csdn.net/es15071848238/article/details/141216773?spm=1001.2014.3001.5501

Harry1326, 2024-08-15:
https://github.com/Harry1326/STM32F103C6T6_HID-bootloader

dmkng, 2025-03-23:
Hello, I've had the same problem long time ago (#51). Typical Blue Pill board uses STM32F103C6T6 (32MB RAM, 10MB Flash) but this bootloader supports only STM32F103C8T6 (64MB RAM, 20MB Flash), I created a fork where I roughly fixed this issue creating new target "bluepill". Here's my fork: https://github.com/dmkng/STM32_HID_Bootloader. Note that the HID bootloader only works with stm32duino/Arduino_Core_STM32, not with rogerclarkmelbourne/Arduino_STM32, but I will try to fix it in my fork soon.