Serasidis/STM32_HID_Bootloader #64: USB init failing for new bluepills

f3sty, 2023-10-07:
I've recently bought two different lots of bluepills that don't work with this bootloader but they work fine with the STM32duino-bootloader, and user space USB works too.
The chips are marked as STM32F103C8T6 but I'm sure they're fake just like most of the bluepills I've bought over the years. At first I thought I'd just received a bad batch, but it looks like many vendors are selling this new clone variant because I then got the same type from another seller :(

They symptoms seem to be they're just running too.... fast? On a 'good' bluepill PC13 LED flashes at ~20hz after burning the bootloader, but on these new ones it flashes at 60hz.
The good bluepill receives its first SOF ~150ms after the start of enumeration, whereas the new bad one receives it 1ms after starting to wake the host. 
But with the STM32duino-bootloader flashed on the new bluepill instead, it too takes ~150ms too. 

With STM32_HID_Bootloader both bluepills receive the SETUP and DATA0 packets from the host about 40-50ms after the first SOF, but these new bluepills never send an ACK (or any response to the SETUP).
The host retries the SETUP transaction two more times without an answer, then tries 3 more times 400ms later.

Seeing as these bluepills work fine with other USB bootloaders the hardware is obviously capable of establishing a USB connection, which leads me to think there might be something in the initialization that they don't like?
Any help would be appreciated, I've got a dozen of these now and I assume there's lot of other people receiving them too.

f3sty, 2023-10-09:
bootsector's original stm32-hid-bootloader also works properly on these new bluepills.

reepotah, 2023-10-23:
Having the same issue with Bluepills I've got few months ago and just unpacked. They get "Unknown USB Device (Device Descriptor Request Failed)" error in a couple of seconds after connecting it to PC.
They look like this:
[image]
[image]
LED on PC13, but it's a red led, same color as the power led.

I don't know how to check USB timings and all that, but my pills are blinking at 27Hz after flashing the bootloader.

One of them runs STLink v2 firmware just fine, also I tried STM32duino-bootloader and it also gets initialized properly.

f3sty, 2023-10-29:
[pulseview_usb.zip](https://github.com/Serasidis/STM32_HID_Bootloader/files/13199348/pulseview_usb.zip)
Here are a pair of pulseview captures of the USB init - one from a bluepill I bought a year or so ago (working), and the other from a recently purchased bluepill (non-working).  
As much as it looks like a hardware issue, these new bluepills work fine with most other USB bootloaders and USB stacks.

onolox, 2024-01-05:
I have a stm32F103CBT6 that have a problem similar to the OP, after flashing the green led stays on, it doesn't flash like it should. After plugging it into USB it shows in windows an problematic device under USB devices.

dylansison, 2024-01-21:
> I have a stm32F103CBT6 that have a problem similar to the OP, after flashing the green led stays on, it doesn't flash like it should (pressing reset or when plugged in USB green led off).
> 
> After plugging it into USB it shows in windows an problematic device under USB devices.

same issues here, and yes other usb bootloaders work just not this one

aguaviva, 2024-04-29:
> same issues here, and yes other usb bootloaders work just not this one

Which bootloaders work fine?

agusting21, 2024-06-14:
@aguaviva  this one works for me https://github.com/rogerclarkmelbourne/STM32duino-bootloader/tree/master/binaries

arthurfprecht, 2024-08-08:
I have the same issue, bought two blue pill boards in different occasions, from different sellers, both with the Device Descriptor Request Failed Error. Similarly to what pelople above found, they work with the STM32duino Bootloader, and if I use the STLink to program them and use CDC, the USB also works just fine ?! 
The funny thing is that they pass the tests in the following tool: https://mecrisp-stellaris-folkdoc.sourceforge.io/bluepill-diags-v1.640.html 
So that mean they are compatible even bug-wise in some things, but not in the USB hardware. That's sad :(

Harry1326, 2024-08-15:
please refer to:  https://blog.csdn.net/es15071848238/article/details/141216773?spm=1001.2014.3001.5501

Harry1326, 2024-08-15:
https://github.com/Harry1326/STM32F103C6T6_HID-bootloader

Harry1326, 2024-08-15:
> Having the same issue with Bluepills I've got few months ago and just unpacked. They get "Unknown USB Device (Device Descriptor Request Failed)" error in a couple of seconds after connecting it to PC. They look like this: ![PXL_20231023_155433814 MV](https://private-user-images.githubusercontent.com/45051999/277349214-c5962384-dc5a-4f5a-af7b-e677f3369472.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjM3MDI2MTYsIm5iZiI6MTcyMzcwMjMxNiwicGF0aCI6Ii80NTA1MTk5OS8yNzczNDkyMTQtYzU5NjIzODQtZGM1YS00ZjVhLWFmN2ItZTY3N2YzMzY5NDcyLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MTUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODE1VDA2MTE1NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWRhZjMwNGVmMzBiMTJjMmNmYWZjNzNmYTU0YmQ5MmJmMGIwNmE3Nzk5MDc1MmE5N2FkOTdhZGZhOTBmMTgzODImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.J6xEmDar6ftmS8GqG4yljmYatf1JrHSVejY8khLGc3o) ![PXL_20231023_155454008 MV](https://private-user-images.githubusercontent.com/45051999/277349257-67a61ac6-2a75-4934-baf1-a7b26d3b8900.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjM3MDI2MTYsIm5iZiI6MTcyMzcwMjMxNiwicGF0aCI6Ii80NTA1MTk5OS8yNzczNDkyNTctNjdhNjFhYzYtMmE3NS00OTM0LWJhZjEtYTdiMjZkM2I4OTAwLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MTUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODE1VDA2MTE1NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTBlNmVlNDQ0MWNmN2FjNjVmNWExYzI3NWNkZWE3NjZlNDFlY2I2NTJkNDUyNTdlMjM3ODNiZTJjMjc1NGFkZjAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.OI3f4AmjPKCvyNokbSfFmIxsclgv7zjW5WWj6_OrMnA) LED on PC13, but it's a red led, same color as the power led.
> 
> I don't know how to check USB timings and all that, but my pills are blinking at 27Hz after flashing the bootloader.
> 
> One of them runs STLink v2 firmware just fine, also I tried STM32duino-bootloader and it also gets initialized properly.

https://github.com/Harry1326/STM32F103C6T6_HID-bootloader

dylansison, 2024-09-09:
> bootsector's original stm32-hid-bootloader also works properly on these new bluepills.

remembered about this issue, the code for USB for the bootsector one seems to be from the same source except more changes occurred after it split; digging into the changes promptly

dylansison, 2024-09-09:
even the earliest commits of this version have the Device Descriptor Request Failed error, interesting. the first version of this and the current version of the Bootsector one looks quite similar and only have a couple differences, wonder it is something that was cut away early on and did not have much effect on operation of normal stm32s but are very impactful on these ones

bins of earlier commits in this repo show more normal speed pc13 led flickering so that might be a separate thing and not a symptom of what may be causing our USB woes here

dylansison, 2024-09-10:
the bugs have been found (i think... or at least one of these is what caused it)... and were found by others in bootsector/stm32-hid-bootloader -> commit [e3b264f](https://github.com/bootsector/stm32-hid-bootloader/commit/e3b264f789ea0f9dfdb0d52c6aa4fbaaa5118bfd) and [e58113c](https://github.com/bootsector/stm32-hid-bootloader/commit/e58113cdc8ca122f92f4751097271444aaf538cb)

more to do though because this version is very different now from bootsector/stm32-hid-bootloader unlike in the old version i threw new edits from the other repo into

dylansison, 2024-09-10:
notes for today:

started bisecting (TIL what git bisect is) and continued experimenting

the change made in bootsector [e58113c](https://github.com/bootsector/stm32-hid-bootloader/commit/e58113cdc8ca122f92f4751097271444aaf538cb) fixes the problems in the earlier versions around serasidis [34d7d73](https://github.com/dylansison/STM32_HID_Bootloader/commit/34d7d7338538097545a49d4ae22417988fb6497e)

[commit 79c474e](https://github.com/dylansison/STM32_HID_Bootloader/commit/79c474e7e36db303623b4cbfbc39cd0f5c9505cf) is the first bad commit

dylansison, 2024-09-11:
The function that was replaced in [commit 79c474e](https://github.com/dylansison/STM32_HID_Bootloader/commit/79c474e7e36db303623b4cbfbc39cd0f5c9505cf) reset the RCC register which after looking at it in STM32CubeProgrammer has a tendency to reset to unexpected values, at least on my blue pills. While I was googling stuff regarding this SystemInit() function. I read some posts about how resetting all of that wasn't necessary since they would be set to the default values on reset so i suppose real STM32F103s do that whilst ours dont

arthurfprecht, 2024-09-14:
@dylansison Awesome work! Please let me know if I can help in any way, I have one of these boards, the debugger and so on. I will try to follow your investigation and see the behaviour on my board here.

dylansison, 2024-09-15:
hi! it would be a big help if you could figure out what exactly in the code fixes the problem and how the code can be cut down to fit back into >=2048bytes

also wonder if this is actually consistent across both of our boards or if the USB init problems have even more causes depending on the chip

arthurfprecht, 2024-09-16:
> https://github.com/Harry1326/STM32F103C6T6_HID-bootloader

Hi @Harry1326, I tested the bootloader on my board (a F103C8) and unfortunately it does not work either, failing enumeration too 😢.

Harry1326, 2024-09-20:
Hi，@arthurfprecht ， if you test F103C8，you just need to burn the repo of this origin bin file:https://github.com/Serasidis/STM32_HID_Bootloader/releases/download/2.2.2/stm32_binaries.zip, i have tested this file ok with F103C8   ,if you test F103C6  you can use the bin file  of my: https://github.com/Harry1326/STM32F103C6T6_HID-bootloader

Harry1326, 2024-09-20:
you can refer to my blog: https://blog.csdn.net/es15071848238/article/details/141216773?spm=1001.2014.3001.5501
use STLink or daplink burn this boot:
[image]

dmkng, 2025-03-23:
Hello, I've had the same problem long time ago (#51). Typical Blue Pill board uses STM32F103C6T6 (32MB RAM, 10MB Flash, even when marked as C8T6) but this bootloader supports only legitimate STM32F103C8T6 (64MB RAM, 20MB Flash), I created a fork where I roughly fixed this issue creating new target "bluepill". Here's my fork: https://github.com/dmkng/STM32_HID_Bootloader. Note that the HID bootloader only works with stm32duino/Arduino_Core_STM32, not with rogerclarkmelbourne/Arduino_STM32, but I will try to fix it in my fork soon.